libcsp/libcsp #839: Fix the null reference vulnerability.

QiuYitai, 2025-04-18:
Hello,
Our team has recently been conducting research on a null-pointer-dereference (NPD) vulnerability detection tool and used it to scan libcsp(the version on the master branch). After a manual review, we have identified some potentially vulnerable code snippets that may lead to null-pointer-dereference bugs. 
The NULL Dereference vulnerability happens in `void csp_kiss_rx()`, `interfaces/csp_if_kiss.c`
How the NULL Pointer Dereference happens:
1. When `ifdata->rx_packet == NULL` and `ifdata->rx_mode == KISS_MODE_STARTED`
2. NULL dereference of variable `ifdata->rx_packet` happens at `ifdata->rx_packet->frame_begin[ifdata->rx_length++] = inputbyte;`
```
void csp_kiss_rx(csp_iface_t * iface, const uint8_t * buf, size_t len, void * pxTaskWoken) {
{
    csp_kiss_interface_data_t * ifdata = iface->interface_data;
    while (len--) {
        uint8_t inputbyte = *buf++;\
=>      if (ifdata->rx_packet != NULL && &ifdata->rx_packet->frame_begin[ifdata->rx_length] >= &ifdata->rx_packet->data[sizeof(ifdata->rx_packet->data)]) {
            ......
        }
        switch (ifdata->rx_mode) {
            ......
            case KISS_MODE_STARTED:
                if (inputbyte == FESC) {......}//false
                if (inputbyte == FEND) {......}//false
                if (ifdata->rx_first) {......}//false
=>              ifdata->rx_packet->frame_begin[ifdata->rx_length++] = inputbyte;
                break;
            ......
        }
    }
 }
```

moonlight83340, 2025-04-18:
Hi, thank you for your report.
By master branch you mean develop branch I guess ?

moonlight83340, 2025-04-18:
Thanks a lot for your report regarding the potential NULL pointer dereference in csp_kiss_rx().

I’ve reviewed the issue and pushed a fix that adds the necessary checks in both KISS_MODE_STARTED and KISS_MODE_ESCAPED states to prevent crashes and increment the appropriate error counters to help with debugging.

While I believe these situations should not normally occur in practice, it's always a good idea to add safeguards just in case.

Thanks again for helping improve libcsp’s robustness!

QiuYitai, 2025-04-25:
Thank you for confirming the NULL pointer dereference vulnerability. To help users track and mitigate this issue, could you please request a CVE ID on [CVE Request Website](https://cveform.mitre.org/).
If you're unable to file this, I can help to submit the request.

PR #840 (2025-07-16): interfaces: kiss: Prevent possible NULL dereference
Add defensive checks in csp_kiss_rx() to avoid potential crashes due to NULL pointer dereference when rx_packet is unexpectedly NULL.

Both KISS_MODE_STARTED and KISS_MODE_ESCAPED now verify that rx_packet is valid before writing to it.

In case of invalid state, rx_error or drop counters are incremented accordingly to aid debugging.

These changes improve the robustness of the KISS interface when dealing with malformed or corrupted input streams.

Fix #839
closed 2025-07-16 by yashi