libcsp/libcsp #843: Embedded Linux Use Case

ledsled, 2025-04-22:
## New to libcsp

We have a physical device and I am trying to communicate over the canbus.  I don't quite understand the Use Case for libcsp.  The vendor documents the interface in terms of a csp protocol.  First thing I am trying is simply using the library to produce can traffic and I am getting traffic.  This simple snippet does produce traffic:

```
    /* open */
    ret = csp_can_socketcan_open_and_add_interface(DEVICE_NAME, CSP_IF_CAN_DEFAULT_NAME, cnode, 1000000, true, &iface);
    if (ret != CSP_ERR_NONE) {
        csp_print("failed to open: %d\n", ret);
        return 1;
    }
    iface->is_default = 1;

    /* connect */
    conn = csp_connect(CSP_PRIO_NORM, snode, port, 1000, CSP_O_CRC32);
    if (conn == NULL) {
        csp_print("Connection failed\n");
        return 1;
    }

    // Get CSP Version
    uint8_t get_payload[16] = {0x05, 0x00, 0xcd, 0x10, 0x40, 0x00, 0x00, 0x18};

    /* prepare data */
    packet = csp_buffer_get_always();
    memcpy(packet->data, get_payload, 8);
    packet->length = 8;

    /* send */
    csp_send(conn, packet);

    while (1) {

        /* Read incoming frame */
        packet = csp_read(conn, 1000);
        if (packet == NULL) {
            break;
        }

        /* We have a reply, ensure data is 0 (zero) terminated */
        const unsigned int length = (packet->length < sizeof(packet->data)) ? packet->length : (sizeof(packet->data) - 1);
        packet->data[length] = 0;
        csp_print("%s", packet->data);

        /* Each packet from csp_read must to be freed by user */
        csp_buffer_free(packet);
    }
 ```
I see can traffic like this:
```
99999 | 10030782 | 00 3C A4 81 05 00 CD 10 .<......
99999 | 10030785 | 40 00 00 18 75 C5 44 18 @...u.D.
99999 | 1001EC02 | 00 61 22 81 06 80 CD 10 .a".....
99999 | 1001EC05 | 40 00 00 02 20 53 22 D2 @... S".
```
That looks correct.  

**First problem** is, I never get anything for a response.  Actually, this might be all I needed if I could field responses.

**Second issue**, I don't really understand the proper topology that I should use.  I have an embedded Linux system and I would love to build payloads for the various commands and pulling telemetry per the ICD but for whatever reason, the libcsp components that would need to be in place haven't lined up in my mind.  Do I build these messages up and send them through the server?  The stock server is spitting out info like this but when I send can traffic through the server, and it is reaching it, it doesn't behave as nicely as when I send it directly using that first code block.  I don't get responses in either case in the code but do see it in the cansniffer.
```
INIT CAN: device: [can0], bitrate: 1000000, promisc: 1
Connection table
[00 0x7fff4ec738a0] S:0, 0 -> 0, 0 -> 0 (17) fl 0
[01 0x7fff4ec739b8] S:0, 0 -> 0, 0 -> 0 (18) fl 0
[02 0x7fff4ec73ad0] S:0, 0 -> 0, 0 -> 0 (19) fl 0
[03 0x7fff4ec73be8] S:0, 0 -> 0, 0 -> 0 (20) fl 0
[04 0x7fff4ec73d00] S:0, 0 -> 0, 0 -> 0 (21) fl 0
[05 0x7fff4ec73e18] S:0, 0 -> 0, 0 -> 0 (22) fl 0
[06 0x7fff4ec73f30] S:0, 0 -> 0, 0 -> 0 (23) fl 0
[07 0x7fff4ec74048] S:0, 0 -> 0, 0 -> 0 (24) fl 0
Interfaces
LOOP       addr: 0 netmask: 14 dfl: 0
           tx: 00000 rx: 00000 txe: 00000 rxe: 00000
           drop: 00000 autherr: 00000 frame: 00000
           txb: 0 (0B) rxb: 0 (0B) 

CAN        addr: 1 netmask: 0 dfl: 1
           tx: 00000 rx: 00000 txe: 00000 rxe: 00000
           drop: 00000 autherr: 00000 frame: 00000
           txb: 0 (0B) rxb: 0 (0B) 

Server task started
Packet received on SERVER_PORT: 
Packet received on SERVER_PORT: 
```

## Environment Details

Provide information about your environment, including:
- Build OS: 6.6.74+rpt-rpi-2712 
- Target OS: Raspberry Pi Linux

## Logs or console output from the very beginning to the end

- [x] I've searched in "Issues" for similar issues, both open and closed.
- [x] I've used GitHub Flavored Markdown (GFM) to format my text.
- [x] I've avoided using screenshots for console output or logs
- [x] I've included all relevant details and steps to reproduce the issue.
- [x] I've provided information about my environment.
- [x] I've attached necessary logs or code snippet for minimal reproducible example.

moonlight83340, 2025-04-23:
Hi,
Can you share your full code used, and parameters ?
I'm not sure I understand but you seems to use the client and server example.
And you would like a answer from the server ?

ledsled, 2025-04-23:
Hey Moonlight,

**What version of libcsp are you using ?**
I'm doing testing on a vendor supplied Raspberry Pi and this is the version they are using:
    _commit cd77e2164441686cd10fd6b4a9279d70b8e11233 (HEAD)
    Merge: 99c103e 7a24dad
    Author: Johan De Claville Christiansen <johan@claville.dk>_

For our purposes, I intend to use the latest but I wanted to get something functioning on a platform that already has a working application.  Our system might have other terms in the equation.

**I'm not sure I understand but you seems to use the client and server example.**

This is part of my question, since I don't quite understand the proper Use Case, I'm not sure if I should use a client/server topology or go connection-less.  

Suppose we have a Linux system and I want to generated traffic to a physical device communicating via CAN Bus, what is the suggested topology? 

**And you would like a answer from the server ?**

Again, my ignorance on full display, but yes.  If I want to pull telemetry from the instrument, how?  I generate a 'get' and in that example above, it looks like the instrument  sees the request and responds with the correct answer, but I only know that by looking at the capturing CAN log.

**Can you share your full code used, and parameters ?**

I have two simple programs where I've tried to pare down the problem.  One does a csp_init(), csp_can_socketcan_open_and_add_interface(), csp_connect(), and csp_send() with subsequent csp_read().  The other uses a csp_transaction_w_opts() without the csp_connect().  Both can hit the server and both can hit the instrument but it is only when I identify the 'server_port' with the actual csp ID that the monitored canbus traffic ever looks correct.

Here is the code for the first program and I run it it with:
_./canbus_send 24 15 10_    (this will generate canbus traffic that will produce that 'correct' can log in my first post. And I can generate some set traffic too (but longer payloads I still have issues)).
./canbus_send 1 2 10  (this will generate canbus traffic that 'hits' the server.  The message from the server indicates as such.  The issue here is I still do not get responses and the instrument is seeing and responding just not correctly.) 

The source code for the server is simply the 'stock' unmodified csp_server.c and csp_server_posix.c.  
```

#include <csp/csp.h>
#include <csp/csp_debug.h>
#include <csp/drivers/can_socketcan.h>
#include <stdio.h>
#include <stdlib.h>

#define SERVER_ADDR 1
#define SERVER_PORT 10

#define CLIENT_ADDR 2
#define DEVICE_NAME "can0"

extern csp_conf_t csp_conf;

int main(int argc, char * argv[])
{
    csp_iface_t  * iface;
    csp_conn_t   * conn;
    csp_packet_t * packet;
    int snode, cnode, port;
    int ret;

    if (argc < 4)
    {
       printf ("arguments: server_node client_node port\n");
       return -1;
    }
    else
    {
       snode = atoi(argv[1]);
       cnode = atoi(argv[2]);
       port  = atoi(argv[3]);
    }

    printf ("snode %d\n", snode);
    printf ("cnode %d\n", cnode);
    printf ("port  %d\n",  port);
    printf ("\n");

    csp_conf.version = 2;

    /* init */
    csp_init();

    /* open */
    ret = csp_can_socketcan_open_and_add_interface(DEVICE_NAME, CSP_IF_CAN_DEFAULT_NAME, cnode, 1000000, true, &iface);
    if (ret != CSP_ERR_NONE) {
        csp_print("failed to open: %d\n", ret);
        return 1;
    }
    iface->is_default = 1;
    /* connect */
    conn = csp_connect(CSP_PRIO_NORM, snode, port, 1000, CSP_O_CRC32);
    if (conn == NULL) {
        csp_print("Connection failed\n");
        return 1;
    }

    // Get RPM with node = 24
    //uint8_t get_payload[16]  = {0x05, 0x00, 0xcd, 0x8f, 0x40, 0x00, 0x00, 0x18};
    uint8_t get_payload[16] = {0x05, 0x00, 0xcd, 0x10, 0x40, 0x00, 0x00, 0x18};

    /* prepare data */
    packet = csp_buffer_get_always();
    memcpy(packet->data, get_payload, 8);
    packet->length = 8;

    /* send */
    csp_send(conn, packet);

    while (1) {

        /* Read incoming frame */
        packet = csp_read(conn, 1000);
        if (packet == NULL) {
            break;
        }

        /* We have a reply, ensure data is 0 (zero) termianted */
        const unsigned int length = (packet->length < sizeof(packet->data)) ? packet->length : (sizeof(packet->data) - 1);
        packet->data[length] = 0;
        csp_print("%s", packet->data);

        /* Each packet from csp_read must to be freed by user */
        csp_buffer_free(packet);
    }

    /* close */
    csp_buffer_free(packet);
    csp_close(conn);

    return 0;
}
```

moonlight83340, 2025-04-23:
> > When you send a packet with CSP, it might seem like you're expecting a response, but CSP doesn't handle that internally. If you want to send a reply upon receiving a packet, you need to implement that logic yourself.
> 
> I'm going to have a look at the client/server you provided but let me clarify here: If I send a command to an actual instrument, it will generate a canbus reply. I'd like to capture that.
> 
> My naive understanding is that libcsp is going to help me manage longer payloads and generation of canbus headers. Since there are payloads longers than 8 bytes, I need it. And I think the notion of a server would facilitate a scheme where multiple clients could be producing traffic.
> 
> Now back to your comment "you need to implement that yourself", this is where I'm falling down. The instrument is producing a canbus reply and will the server capture that and I need to spit that back in a reply?

I'm not sure I have fully understand so, I will first answer here.

> My naive understanding is that libcsp is going to help me manage longer payloads and generation of canbus headers. Since there are payloads longers than 8 bytes, I need it. And I think the notion of a server would facilitate a scheme where multiple clients could be producing traffic.

Yes it will. csp will help you to send payload throw the can interface for example, for you so you don't have to cut your payload in 8 bits yourself to reconstruct it on the server part.
So here you understand it well.

> Now back to your comment "you need to implement that yourself", this is where I'm falling down. The instrument is producing a canbus reply and will the server capture that and I need to spit that back in a reply?

Why I'm telling that ? Because when you use csp you let csp send and receive the payload so if you want csp to send a answer to the client you can do it but it's not implement in the core.
But I'm not sure I understand well when you're speaking about the server answer.

ledsled, 2025-04-23:
Moonlight, 

Thanks for taking the time. It is much appreciated.

> I'm not sure I have fully understand so, I will first answer here.

Let try and clarify this. If I send a CAN packet through libcsp to an instrument, that instrument responds. I want to read that response.  Probably csp_read().  I never get anything back from csp_read().  Ever.  Now if I monitor the CAN traffic (I use cansniffer) I see responses from the instrument but programmatically, nothing.

> Yes it will. csp will help you to send payload throw the can interface for example, for you so you don't have to cut your payload in 8 bits yourself to reconstruct it on the server part. So here you understand it well.

Fantastic.  Honestly, this is why libcsp is appealing to me. 

> Why I'm telling that ? Because when you use csp you let csp send and receive the payload so if you want csp to send a answer to the client you can do it but it's not implement in the core. But I'm not sure I understand well when you're speaking about the server answer.

So if I have client that generates csp packets over CANbus to the server, the server is receiving it.  The stock server emits a message "Packet received on SERVER_PORT".   This is where I'm not understanding what is needed.  The server looks like it starts up and it prints a "Connection table" that does indeed have my instrument listed.  I would think it would forward the packet as there is a router started and I've seen some discussion about "next hop".  But how/what the linkage is still hasn't come clear in my head.  

I am going to look at that part of the code though in the server right now and see if anything slides in to place.

If I can get this topology and Hello World working, I can generate CAN bus traffic to my instrument and field responses from it and all is well with the world.  Well, this part of the world.

> edit: I will take more time tomorrow to answer all your questions, sorry it's a bit late here.

Good for you.  Again, thanks for your time.

ledsled, 2025-04-23:
Looking at the stock server, I see this:

```
                while ((packet = csp_read(conn, 50)) != NULL) {
                        switch (csp_conn_dport(conn)) {
                        case SERVER_PORT:
                                /* Process packet here */
                                csp_print("Packet received on SERVER_PORT: %s\n", (char *) packet->data);
                                csp_buffer_free(packet);
                                ++server_received;
                                break;

                        default:
                                /* Call the default CSP service handler, handle pings, buffer use, etc. */
                                csp_service_handler(packet);
                                break;
                        }
                }
```

So the default handles all the CSP service ports like ping, reset, etc.

The SERVER_PORT then would be where I implement our/my logic?  At this point I'd inspect it and build up a CANbus packet, or simply resend the current CANbus packet to the appropriate CSP ID?  If so, how?  Not using a connection I'd guess.  Also, I would still want to get the responses from the instrument.  Maybe that lives in another thread but I still cannot ever capture my instrument's replies.

moonlight83340, 2025-04-24:
Hi,

I made a simple example using a connection to send a packet and get a reply:
👉 https://github.com/moonlight83340/libcsp_examples

It’s designed to be as simple as possible, just to demonstrate the basics — of course, there are many other possibilities with CSP.

If you just want to make sure your node receives a packet, you could also consider using the RDP option. It will ensure packet delivery if no ACK is received on the client side.

In my example, Node 1 sends a message every second to Node 2, which listens for incoming traffic and replies with a basic ACK message that includes the number of received packets.

As for your question about using connections vs connection-less mode in libcsp:
The best choice depends on your use case.
If you're just sending simple one-shot commands and don't need a full connection, then connection-less might be enough.
But if you need acknowledgements, retries, or two-way communication, using a connection can be safer and more robust.

Let me know if you’d like help adapting the example to your setup!

ledsled, 2025-04-24:
Hi Moonlight,

Thank you so much for the response.  The RDP example is particularly illuminating.  I think once I get past this one issue that example would be very useful.

Here is the problem I'm having though.  The client/server communicating commands/responses isn't an issue for me.  What is an issue is:

If I generate a CAN message to the vendor supplied instrument (I do not have control of the instrument software) I need to be able to capture the response.  For example, this is a command I send:

06235 | 10030382 | 00 1C A4 81 **05 00 CD 10 ........
06235 | 10030385 | 40 00 00 18** 75 C5 44 18 @...u.D.

The bolded portion is the Payload.  The 05 is a 'get' and the '10' identifies the item to retrieve and the '18' is the CSP node id.  This is the response from the instrument with the payload in bold.

99999 | 1000EC42 | 00 61 22 81 **06 80 CD 10 .a".....
99999 | 1000EC45 | 40 00 00 02** 20 53 22 D2 @... S".

And that is a correct response to this particular command.  The 06 is the code for a 'get response' and the '10' identifies the argument and the '02' is the value I asked for (which is CSP version the instrument is using).  My issue is I cannot capture the response from the instrument.  These values I'm showing you I only see because I capture them on a CANbus sniffer.  I can generate CAN traffic but I cannot read the responses.

moonlight83340, 2025-04-24:
Hi,
I may have understand something, I wasn't understanding before.
You're sending the can message throw csp to the instrument, which seems to get it but the instrument then respond itself so yes the libcsp part can't see that...
Do you have any way to get the instrument answer and send it by csp on server side ?

what I have understand :
Client --> send command --> Server
instrument --> response --> Client

But I think it should be : 
CSP Client --> send command --> CSP Server
instrument --> response --> CSP Server --> response --> CSP Client
something like that, csp will be on top of you're can, but if raw can is send whithout csp header, there is no way (that I know) to csp to saw that packet.

moonlight83340, 2025-04-24:
I don't know what instrument you are using but can you get or avoid to directly send the can answer to get a way to pass the payload to csp ? 🤔

ledsled, 2025-04-24:
I think what you're telling me is that I'm not getting responses because it looks like I'm not generating proper CSP traffic.  That is, there is not CSP header.   Is that what you are saying?

But I am using csp_send() to generate traffic.  I would think that would generate a CSP header.

moonlight83340, 2025-04-24:
> I think what you're telling me is that I'm not getting responses because it looks like I'm not generating proper CSP traffic. That is, there is not CSP header. Is that what you are saying?
> 
> But I am using csp_send() to generate traffic. I would think that would generate a CSP header. It actually looks like a CSP header and it has the associated CRC.
> 
> 06235 | 10030382 | **_00 1C A4 81_** **05 00 CD 10 ........ 06235 | 10030385 | 40 00 00 18** _75 C5 44 18_ @...u.D.
> 
> The **_bold italics_** is the CSP header and the **bold** is the payload and the _italics_ is the CRC.

I tougth you didn't get an answer from your instrument because of that.
So if you answer with a correct csp packet from the instument that weird..
I will need more information ,code to understand I guess.

ledsled, 2025-04-24:
Yes.  I guess that is what I am saying.  

The instrument is spitting out what looks like a CSP response too.  Here is the instrument response:

99999 | 1000EC42 | 00 61 22 81 06 80 CD 10 .a".....
99999 | 1000EC45 | 40 00 00 02 20 53 22 D2 @... S".

That '06' is a response to an '05' with the same parameter id '10' and a correct value preceding the CRC of '02'.  

In terms of code, yes, what?  I've tried establishing a connection with the instrument and using csp_send() to send a command and then subsequently try to csp_read() on that connection but do not get anything back.  And yet, that csp_send() produces canbus traffic that looks like it works.  I'm baffled.

That program I posted earlier run with arguments '24  15 10' will produce canbus traffic that looks like it works.  In this case '24' is the instrument CSP id and '10' is the port.  I don't think the '15' matters but the variable cnode gets that value when run.

moonlight83340, 2025-04-24:
> Yes. I guess that is what I am saying.
>
> The instrument is spitting out what looks like a CSP response too. Here is the instrument response:
>
> 99999 | 1000EC42 | 00 61 22 81 06 80 CD 10 .a"..... 99999 [...]

Let me, check again.
What's your client, server address ? Define in the client code and server ?
24 is the client address and 15 the server address ? does that match the server address ? (15 I guess ?)
10 for the port seems okay.

ledsled, 2025-04-24:
If I run the server, it is address 1.  The CSP ID of the instrument is 24.

This is my problem.  If I run that program above without a server, and use arguments '24 2 10' then the generated canbus traffic does indeed look correct.  But my subsequent csp_read() times out.  

If I run the server and direct traffic at the server with arguments '1 2 10' then yes, the server does hit the 'SERVER_PORT' switch case.  But in that case, my client connecting and sending a packet to the server is reading from a connection with the server.

I still fail to grasp where the actual physical hardware instrument response is captured.  That is the problem. Where?  **Where and how is that typically picked up?**

moonlight83340, 2025-04-24:
if you run the program with argument 24 2 10, you will have a client ID of 24 for the instrument and an address of 2 for the server your want to send the packet on the port 10.
Does that seems correct to you ?

ledsled, 2025-04-24:
- Actually, I think '24 2 10' will produce a server ID of 24 and a client ID of 2 and a port of 10. This produces canbus traffic that looks correct in terms of the instrument command/response.  

- I think '1 2 10' will produce a server ID of 1 and a client ID of 2 and a port of 10.  This reaches the server on the switch statement.  But monitoring the canbus I don't see the correct response from the physical device.  

But my problem is that I just do not see how one could capture the libcsp traffic.  I seem to be able to produce it but cannot ever read it.

moonlight83340, 2025-04-24:
Hey, I think it would really help to understand the full picture here.

Would you mind sharing a quick diagram or even just a description of how your setup is structured? Specifically:

    Who is the client? (your program?)

    Who is the server? (your program again, or something else?)

    Who/what is the instrument?

    Which parts are using libcsp and which aren't?

    And finally, what is the expected direction of the messages?

Right now, it's a bit unclear to me where the request originates and where the response should be captured from libcsp's perspective.

Sorry again, I want to confirm the code.
You're currently using the code you paste before + the csp server code ? 
I will test on my machine your code to check if everything is running correctly first.

ledsled, 2025-04-24:
1) If I run that program I pasted in by itself, no server running, like this:

$ ./canbus_send 24 2 10   

I get canbus traffic like this:

99999 | 10002C62 | 00 61 22 81 06 80 CD 10 .a".....
99999 | 10002C65 | 40 00 00 02 20 53 22 D2 @... S".
99999 | 10030082 | 00 04 A4 81 05 00 CD 10 ........
99999 | 10030085 | 40 00 00 **18** 75 C5 44 18 @...u.D.

NOTE:  The instrument has a CSP id of 24.  In fact that bolded 18 is part of this particular instrument's protocol. 

2) If I run that same program I pasted with the server running, like this:

$ ./csp_server -c can0 -a 1   (in one terminal)
$ ./canbus_send 1 2 10        (in another terminal)

I get canbus traffic like this:

99999 | 00004C22 | 00 60 A1 41 00 00 00 0E .`.A....
99999 | 00004C25 | E4 1F F4 E0             ....
99999 | 00030082 | 00 08 52 81 05 00 CD 10 ..R.....
99999 | 00030085 | 40 00 00 **18** 75 C5 44 18 @...u.D.

NOTE:  In the above log, I don't believe the instrument is responding correctly.  It's certainly different.   The CSP id is still embedded in the payload of the CSP message but I don't think the CAN header or CSP header is set up correctly to reach the instrument in this case.

But, the point is, in NEITHER case am I able to capture a response from the instrument itself.  I cannot seem to figure out how to coerce CSP in to reading or receiving CAN or CSP responses from the hardware device itself.

ledsled, 2025-04-24:
I changed the code a bit to set up 'promiscuous' mode.  I pasted it below.  And when I run it, it is capturing the packet I generate but not the response.

```
pi@veopi19:~/temp $ ./canbus_send 24 2 10
snode 24
cnode 2
port  10

INIT CAN: device: [can0], bitrate: 1000000, promisc: 1
RTNETLINK answers: Operation not permitted
RTNETLINK answers: Operation not permitted
RTNETLINK answers: Operation not permitted
RTNETLINK answers: Operation not permitted
0x05 0x00 0xcd 0x10
```
And here is the output in the log:

```
99999 | 10004C42 | 00 61 22 81 06 80 CD 10 .a".....
99999 | 10004C45 | 40 00 00 02 20 53 22 D2 @... S".
99999 | 10030102 | 00 08 A4 81 05 00 CD 10 ........
99999 | 10030105 | 40 00 00 18 75 C5 44 18 @...u.D.
```
And that actually looks good.  I decomposed the CAN header and the CSP header and they look good.   

Still don't understand why the instrument response isn't captured with the promiscuous mode.  But at least I got something even though it's the packet I generated.

```
#include <csp/csp.h>
#include <csp/csp_debug.h>
#include <csp/drivers/can_socketcan.h>
#include <stdio.h>
#include <stdlib.h>

#define SERVER_ADDR 1
#define SERVER_PORT 10

#define CLIENT_ADDR 2
#define DEVICE_NAME "can0"

extern csp_conf_t csp_conf;

int main(int argc, char * argv[])
{
    csp_iface_t  * iface;
    csp_conn_t   * conn;
    csp_packet_t * packet;
    int snode, cnode, port;
    int ret;

    if (argc < 4)
    {
       printf ("arguments: server_node client_node port\n");
       return -1;
    }
    else
    {
       snode = atoi(argv[1]);
       cnode = atoi(argv[2]);
       port  = atoi(argv[3]);
    }

    printf ("snode %d\n", snode);
    printf ("cnode %d\n", cnode);
    printf ("port  %d\n",  port);
    printf ("\n");

    csp_conf.version = 2;

    /* init */
    csp_init();

    /* open */
    ret = csp_can_socketcan_open_and_add_interface(DEVICE_NAME, CSP_IF_CAN_DEFAULT_NAME, cnode, 1000000, true, &iface);
    if (ret != CSP_ERR_NONE) {
        csp_print("failed to open: %d\n", ret);
        return 1;
    }
    iface->is_default = 1;
    
/* connect */
    conn = csp_connect(CSP_PRIO_NORM, snode, port, CSP_MAX_TIMEOUT, CSP_O_CRC32);
    if (conn == NULL) {
        csp_print("Connection failed\n");
        return 1;
    }

    // Get RPM with node = 24
    //uint8_t get_payload[16]  = {0x05, 0x00, 0xcd, 0x8f, 0x40, 0x00, 0x00, 0x18};
    uint8_t get_payload[16] = {0x05, 0x00, 0xcd, 0x10, 0x40, 0x00, 0x00, 0x18};

    /* prepare data */
    packet = csp_buffer_get_always();
    memcpy(packet->data, get_payload, 8);
    packet->length = 8;

    csp_promisc_enable(10);

    /* send */
    csp_send(conn, packet);

    while (1) {

        /* Read incoming frame */
        //packet = csp_read(conn, CSP_MAX_TIMEOUT);
        packet = csp_promisc_read(CSP_MAX_TIMEOUT);
        if (packet == NULL) {
            break;
        }

        /* We have a reply, ensure data is 0 (zero) termianted */
        const unsigned int length = (packet->length < sizeof(packet->data)) ? packet->length : (sizeof(packet->data) - 1);
        csp_print("0x%02x 0x%02x 0x%02x 0x%02x\n", packet->data[0], packet->data[1], packet->data[2], packet->data[3]);

        /* Each packet from csp_read must to be freed by user */
        csp_buffer_free(packet);
    }

    /* close */
    csp_buffer_free(packet);
    csp_close(conn);

    return 0;
}

```

moonlight83340, 2025-04-25:
Hi! I wanted to ask which reaction wheel you're using in your system. Also, do you know which version of CSP it's using ?

ledsled, 2025-04-25:
Using CSP 2.  Actually that transaction I posted is a 'get csp version' command.  And it responds with '2'.

It's a SpaceInventor part.

moonlight83340, 2025-04-25:
> Using CSP 2. Actually that transaction I posted is a 'get csp version' command. And it responds with '2'.
> 
> It's a SpaceInventor part.

Hey,

I'm wondering if there could be an issue with the source and destination addresses in the packet. To be honest, I'm not entirely sure after looking at your CAN frame.

But if you're able to capture the CSP CAN frame using a CAN sniffer, it might be that the address isn't correctly assigned to you (maybe). I'll think about this more over the weekend. If you could paste me some more logs and confirm the source and destination addresses from your logs, I can check them.

Thanks again!

moonlight83340, 2025-04-28:
> Hi Moonlight,
> 
> Thank you so much for the response. The RDP example is particularly illuminating. I think once I get past this one issue that example would be very useful.
> 
> Here is the problem I'm having though. The client/server communicating commands/responses isn't an issue for me. What is an issue is:
> 
> If I generate a CAN message to the vendor supplied instrument (I do not have control of the instrument software) I need to be able to capture the response. For example, this is a command I send:
> 
> 06235 | 10030382 | 00 1C A4 81 **05 00 CD 10 ........ 06235 | 10030385 | 40 00 00 18** 75 C5 44 18 @...u.D.
> 
> The bolded portion is the Payload. The 05 is a 'get' and the '10' identifies the item to retrieve and the '18' is the CSP node id. This is the response from the instrument with the payload in bold.
> 
> 99999 | 1000EC42 | 00 61 22 81 **06 80 CD 10 .a"..... 99999 | 1000EC45 | 40 00 00 02** 20 53 22 D2 @... S".
> 
> And that is a correct response to this particular command. The 06 is the code for a 'get response' and the '10' identifies the item and the '02' is the value I asked for (which is CSP version the instrument is using). My issue is I cannot capture the response from the instrument. These values I'm showing you I only see because I capture them on a CANbus sniffer. I can generate CAN traffic but I cannot read the responses.

I checked both CAN frames, and everything looks fine in the header. The only thing I'm unsure about is the timestamp.
If it's the same packet exchange and you're sending the command at 06235 but receiving the response at 99999, it seems strange — but I'm not sure yet.
I'll spend more time on it if I can get more logs to better understand what's going on.

moonlight83340, 2025-04-30:
> 99999 | 10002C62 | **00 61 22 81** 06 80 CD 10 .a"..... 99999 | 10002C65 | 40 00 00 02 20 53 22 D2 @... S". 99999 | 10030082 | 00 04 A4 81 05 00 CD 10 ........ 99999 | 10030085 | 40 00 00 18 75 C5 44 18 @...u.D.
> 
> That CSP 2 header there for the response, does that look correct? 0000 0000 0110 0001 0010 0010 1000 0001 Source ID: 0x184 Destination Port: 0x12
> 
> The CAN Header 0x10030082 looks correct with a Dest ID of 0x18 and a Source ID lower 6 bits of 0x01.
> 
> I wonder if that is the issue? The software might be looking for a Source ID in the CSP header of 0x18?

What I can see with those frames is :
First frame will have : 
- Priority of 2
- Destination 1
- Source 24
- Destination Port 18
- Source Port 10
- Flag 1
Second frame will have : 
- Priority of 2
- Destination 24
- Source 1
- Destination Port 10
- Source Port 18
- Flag 1

Does that sound correct to you ?
Look like the packet are correct. It's even responding isn't it ?
Do you have any documentation ? I don't know how the Space Inventor instrument are configured.
So I don't know if the instrument address are looking for some address statically or if there address are also statically set. (Set by default)

moonlight83340, 2025-05-08:
Any advance are or news on your side ?

yashi, 2025-07-16:
@ledsled, please reopen or open new one if you have problems. I'm closing this.